eclipse-threadx/getting-started #69: B-L475E-IOT01A reminder Initializing SNTP client 	FAIL: Unable to resolve DNS for SNTP Server time.google.com (0xa3)

colinvsyuan, 2020-05-22:
azure-rtos/getting-started/B-L475E-IOT01A reminder below error after flash:
Initializing SNTP client
	FAIL: Unable to resolve DNS for SNTP Server time.google.com (0xa3)
ERROR: Failed to run the sntp client

colinvsyuan, 2020-05-22:
resolved by replace board.

ryanwinter, 2020-05-29:
Hi Colin,

It's possible the previous board was running an older wifi firmware version. Are you able to post what's printed out when the board starts up with the GSG firmware?

colinvsyuan, 2020-06-01:
Hi Ryan,
Not sure how to check and flash the wifi firmware? how to check the GSG firmware post? I could try it if you will guide?

ryanwinter, 2020-06-01:
When you run through the guide, you will get the following output on the serial console:
```
Initializing WiFi
	Module: ISM43362-M3G-L44-SPI
	FW Revision: C3.5.2.5.STM
```

I believe FW Revisions older than 3.5.2.5.STM have issues. Are you able to check what the version is?

colinvsyuan, 2020-06-01:
your are right, my wifi firmware is "C3.5.2.3.BETA9"

ryanwinter, 2020-06-03:
Doc fix with https://github.com/azure-rtos/getting-started/pull/76

closed 2020-06-03 by ryanwinter