eclipse-threadx/getting-started #74: Replace CPM with git sub modules

JakeSays, 2020-05-31:
Since this project is tightly coupled to threadx repos it would be appreciated if you would replace the non-standard CPM cmake module with standard git sub modules. CPM is just unnecessary complexity, and makes integrating with existing development workflows (not CI workflows) difficult.

ryanwinter, 2020-06-01:
Hi @JakeSays,

Thanks for the feedback. Could you describe the particular problem you are seeing with integration with existing development workflows?

CPM is a thin wrapper over cmake [FetchContent](https://cmake.org/cmake/help/latest/module/FetchContent.html). Would using the standard FetchContent itself by an acceptable solution for you?

JakeSays, 2020-06-01:
@ryanwinter Actually my issue isn't with CPM per se. The biggest problem is having dependencies located in the build directory. That directory is transient, and having to refresh dependencies constantly is both time consuming and frustrating. It is not uncommon to have several threadx files open in the editor when debugging or studying code.

The typical workflow is to have a `third_party` type directory under the source tree with subdirectories for each dependency. Those directories then either contain whatever is necessary to satisfy the dependency, or when necessary the dependency itself. Usually the git based deps are pulled in as submodules, but in this case you could have cmake files in there that handle CPM, etc.

ryanwinter, 2020-06-02:
Thanks @JakeSays, this makes sense. Let me think about this a bit more.

ryanwinter, 2020-06-05:
Tracking on this branch: https://github.com/azure-rtos/getting-started/tree/user/rywinter/submodule

ryanwinter, 2020-06-05:
Tracking on PR: https://github.com/azure-rtos/getting-started/pull/82

closed 2020-06-09 by ryanwinter