eclipse-threadx/getting-started #78: Compiler errors for ARM GCC 10

TiejunMS, 2020-06-04:
Though it is compiler clean for ARM GCC 9, it would be great to fix the following compiler error reported by ARM GCC 10.
```
/usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: app/CMakeFiles/stm32l4S5_azure_iot.dir/networking.c.obj:(.bss.nx_ip+0x0): multiple definition of `nx_ip'; common/CMakeFiles/app_common.dir/networking.c.obj:(.bss.nx_ip+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: app/CMakeFiles/stm32l4S5_azure_iot.dir/networking.c.obj:(.bss.nx_pool+0x0): multiple definition of `nx_pool'; common/CMakeFiles/app_common.dir/networking.c.obj:(.bss.nx_pool+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: app/CMakeFiles/stm32l4S5_azure_iot.dir/networking.c.obj:(.bss.nx_dns_client+0x0): multiple definition of `nx_dns_client'; common/CMakeFiles/app_common.dir/networking.c.obj:(.bss.nx_dns_client+0x0): first defined here
```

ryanwinter, 2020-06-04:
Thanks @TiejunMS, I think this fixes the problem: https://github.com/azure-rtos/getting-started/pull/79

closed 2020-06-04 by ryanwinter

TiejunMS, 2020-06-04:
Verified. It is fixed. Thanks!