eclipse-threadx/getting-started #92: Compile error

rploeg, 2020-07-06:
Hi,

I just followed the guidance for the STM RTOS Device, but when I try to build the new image I get the following error:

`C:\repo\getting-started\STMicroelectronics\STM32L4\tools>rebuild.bat
-- Using ARM GCC from path = C:/Program Files (x86)/GNU Tools Arm Embedded/9 2019-q4-major/bin
-- Using ARM GCC from path = C:/Program Files (x86)/GNU Tools Arm Embedded/9 2019-q4-major/bin
-- The C compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Tools Arm Embedded/9 2019-q4-major/bin/arm-none-eabi-gcc.exe
-- Check for working C compiler: C:/Program Files (x86)/GNU Tools Arm Embedded/9 2019-q4-major/bin/arm-none-eabi-gcc.exe
-- Check for working C compiler: C:/Program Files (x86)/GNU Tools Arm Embedded/9 2019-q4-major/bin/arm-none-eabi-gcc.exe - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Using toolchain file: C:/repo/getting-started/cmake/arm-gcc-cortex-m4.cmake
CMake Error at lib/CMakeLists.txt:18 (add_subdirectory):
  The source directory

    C:/repo/getting-started/core/lib/threadx

  does not contain a CMakeLists.txt file.

-- Configuring incomplete, errors occurred!
See also "C:/repo/getting-started/STMicroelectronics/STM32L4/build/CMakeFiles/CMakeOutput.log".
See also "C:/repo/getting-started/STMicroelectronics/STM32L4/build/CMakeFiles/CMakeError.log".
ninja: error: loading 'build.ninja': The system cannot find the file specified.

C:\repo\getting-started\STMicroelectronics\STM32L4\tools>`

I am running on latest version of Windows 10 and installed all the software that was required. I have there the CMakeError output:

`Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: C:/Program Files (x86)/GNU Tools Arm Embedded/9 2019-q4-major/bin/arm-none-eabi-gcc.exe 
Build flags: -mthumb;-mcpu=cortex-m4;-mfloat-abi=hard;-mfpu=fpv4-sp-d16;--specs=nano.specs;-u;_printf_float;-ffunction-sections;-fdata-sections;-Wunused;-Wuninitialized;-Wall
Id flags:  

The output was:
1
c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard\libc_nano.a(lib_a-exit.o): in function `exit':
exit.c:(.text.exit+0x1a): undefined reference to `_exit'
c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard\libc_nano.a(lib_a-sbrkr.o): in function `_sbrk_r':
sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `_sbrk'
collect2.exe: error: ld returned 1 exit status`

Any idea what I am doing wrong?

rploeg, 2020-07-06:
My clone didn't worked correctly, so some files where missing....

closed 2020-07-06 by rploeg

ryanwinter, 2020-07-06:
Hi @rploeg, glad you resolved this. Do you know why the clone failed?