eclipse-threadx/getting-started #97: compile error on B-L475E-IOT01A board

zzzh, 2020-07-20:
Hello,

I was following this guide to compile it for B-L475E-IOT01A board, but got errors like:

../lib/netx_driver/src/es_wifi.h:253:22: error: field 'IO_Init' declared as a function
  253 |   IO_Init_Func       IO_Init;

the code is: 
typedef struct {
  IO_Init_Func       IO_Init;
  ...
} ES_WIFI_IO_t;

IO_Init_Func is defined as:
typedef int8_t *IO_Init_Func(uint16_t );

It looks good to me, any ideas on this error?

ryanwinter, 2020-07-20:
Hi @zzzh,

What compiler version are you using?

Also can you paste more of the error?

Thanks

zzzh, 2020-07-20:
Hi, @ryanwinter，

The compiler is the one in get-toolchain.bat: gcc-arm-none-eabi-9-2019-q4-major-win32-sha2.exe
only cmake is different, which is 3.17.3

All errors by now is in same type, and all in ES_WIFI_IO_t struct, so the one here should be enough.

Thanks for your reply.

zzzh, 2020-07-20:
Hi, @ryanwinter，

Sorry, I realized that I'm actually using programs extracted from gcc-arm-none-eabi-9-2019-q4-major-win32.zip package, that should be fine, right?

ryanwinter, 2020-07-21:
Hi,

This error isn't reproducible for me so I'll have to investigate a bit further. Couple of things to test:
1. Can you delete and reclone the repo, make sure you clone recursive? Maybe I file was corrupted on the clone.
2. If the file path is quite deep, can you try and checkout to a shallower path? I know that some compiler have issues with deep file paths.

Ryan

zzzh, 2020-07-21:
Sure, I'll do that tomorrow.

Thanks.

zzzh, 2020-07-22:
Hi, @ryanwinter，

I did the 2 steps as you recommended, it compile perfectly now. 

thanks for the help.

Bob

closed 2020-07-22 by zzzh