eclipse-threadx/getting-started #125: ST's netxduo of wifi version failure in DPS

ccli8, 2020-08-18:
Hi,

I am Nuvoton's engineer and on porting azure iot onto Nuvoton's NuMaker-IoT-M487 board, which is Cortex-M4 MCU and has one on-board ESP8266 wifi module. Because ESP8266 WiFi module has its own stack, I follow [ST's reduced netxduo for wifi](https://github.com/azure-rtos/getting-started/tree/master/STMicroelectronics/STM32L4_L4%2B/lib/netxduo) to substitute for the standard one. Adding ESP8266 wifi driver, it can work well with this netxduo of wifi version.

To connect with azure iot, I port the sample [sample_azure_iot_embedded_sdk](https://github.com/azure-rtos/azure-iot-preview/tree/master/samples/sample_azure_iot_embedded_sdk). Without DPS enabled, I can successfully connect with azure iot hub. But after enabling DPS, it fails because the netxduo of wifi version disables the `nx_tcp_socket_establish_notify(...)`. 

https://github.com/azure-rtos/getting-started/blob/829122217a643ce60d0b468c71358fcb9522aa38/STMicroelectronics/STM32L4_L4%2B/lib/netxduo/src/nx_tcp_socket_establish_notify.c#L78

Do you have plan to enhance the netxduo of wifi version to support use case like DPS here?

TiejunMS, 2020-08-18:
@ccli8 , it is limited by the WIFI driver. We are not able to support it. To avoid this issue, you can specify a timeout in connecting APIs.

For ESP8266, it is possible to port a driver with full netxduo features. Though we don't have a plan to port it in the near future, I heard it is working from partner.

ccli8, 2020-08-18:
@TiejunMS Thanks for your reply. Borrowing this thread, for communication module with its own stack like ESP8266 here, do you have another plan to support it, or only follow ST's netxduo for wifi version currently?

TiejunMS, 2020-08-18:
@yuxin-azrtos , could you answer this question?

yuxin-azrtos, 2020-08-27:
@ccli8 We are looking at supporting a wide range of communication modules, including official support for the ESP8266. We also plan to add more documentation on how to support new communication modules in the future.

I am closing this issue.  Feel free to reopen it if you have further questions.

closed 2020-08-28 by ryanwinter