eclipse-threadx/getting-started #127: Disable interrupt in ST's netx wifi driver?

ccli8, 2020-08-20:
I have a question with protection of ST's netx wifi driver. In https://github.com/azure-rtos/getting-started/blob/master/STMicroelectronics/STM32L4_L4%2B/lib/netx_driver/src/nx_wifi.c,

`nx_wifi_` APIs have been protected with:

https://github.com/azure-rtos/getting-started/blob/829122217a643ce60d0b468c71358fcb9522aa38/STMicroelectronics/STM32L4_L4%2B/lib/netx_driver/src/nx_wifi.c#L228-L229

Then why needs disabling interrupt `TX_DISABLE`

https://github.com/azure-rtos/getting-started/blob/829122217a643ce60d0b468c71358fcb9522aa38/STMicroelectronics/STM32L4_L4%2B/lib/netx_driver/src/nx_wifi.c#L276-L277
`

ryanwinter, 2020-08-20:
@yuxin-azrtos,

Are you able to expand on @ccli8's question?

bo-ms, 2020-08-21:
Hi @ccli8, thanks for bring this up, yes, it is not necessary to disable interrupt again here after getting mutex.

ryanwinter, 2020-09-01:
Hi @ccli8, we are still investigating this question.

ryanwinter, 2020-09-15:
Hi @ccli8,

We are still looking into this, as this has no customer impact, its a lower priority right now.

Thanks

yuxin-azrtos, 2020-09-26:
Hi @ccli8 just so you know this issue is in our tracking system, and we will review and respond to you.  Thank you for your understanding.

closed 2020-09-29 by ryanwinter