eclipse-threadx/getting-started #135: Build Failure

surajsridhar, 2020-09-08:
Hello, 

I am using the STMircro BL475E board. 

I have executed the batch file that installs all the dependencies latest revision. But when I run the revbuild.bat file in adminstrator mode, the build fails and I get CMake error. 

I have the latest version of CMake installed. Verified it with the cmake --version command. Can you please let me know how to solve the issue. Attached is the error log. 

[Error_log.txt](https://github.com/azure-rtos/getting-started/files/5190407/Error_log.txt)

ryanwinter, 2020-09-08:
Hi @surajsridhar,

The build can't find the arm compiler so is failing. It also seems you hit upon a bug in the toolchain file where it is trying to resolve the compiler through a visual studio installation.

You can either run "get_toolchain.bat" to setup the gcc arm compiler, or you can download it directly from Arm itself and add it to your system path.

surajsridhar, 2020-09-08:
[image]

This is the output i get when i run the get-toolchain.bat file as an adminstrator. If I am running the wrong file, can you point me to the correct directory.

ryanwinter, 2020-09-08:
This is the right file, looks like there are some issues with gcc and termite, let me look into this and get back to you.

ryanwinter, 2020-09-08:
Can you look in the **C:\Users\sridhars\AppData\Local\Temp** directory. 

Do you have a **termite-3.4.exe** and **gcc-arm-none-eabi-9-2019-q4-major-win32-sha2.exe** file in there?

surajsridhar, 2020-09-08:
Hello, 

I did look into those directories and Termite and gcc-arm are not present. I have termite in my C:\ProgramFiles9x86)\Termite and I assume it is installed. 

Also, based on the error I got on running the rebuild.bat file in adminstrator mode, I did download the arm-none-eabi-gcc.exe from the ARM website, ran the installer until the command prompt opened, closed it  and set the path manually as mentioned in the online document - https://mynewt.apache.org/latest/get_started/native_install/cross_tools.html#installing-the-arm-toolchain-for-windows

ryanwinter, 2020-09-08:
I haven't quite figured out while the download is failing in for the arm compiler. Is it possible that there is a virus scanner running that is quarentining the files?

surajsridhar, 2020-09-08:
I have symantec encryption present in my laptop. But my colleagues who have the same thing, have executed the project without any issues. We have the same security measures.

ryanwinter, 2020-09-08:
If you are unblocked for now, I'll close this item.

I will look into adding manual instruction for installing the tools in the future.

surajsridhar, 2020-09-08:
I am still facing issues. You can close the ticket but it will be unresolved.

ryanwinter, 2020-09-08:
Sorry I misunderstood, you mentioned above that you manually install the gcc arm toolkit

If you download it from here:
https://developer.arm.com/-/media/Files/downloads/gnu-rm/9-2019q4/gcc-arm-none-eabi-9-2019-q4-major-win32-sha2.exe

Then add the bin folder into your PATH, then it will be picked up automatically.

surajsridhar, 2020-09-09:
Hello, 

I have re verified the path of the GCC compiler and other software requirements. I repeated the steps using the STM BL475E IoT board. 

But when I run the rebuild.bat file in the adminstrator mode i am still getting issues, I have verified that I have the latest CMake (CMake --version) and Ninja(ninja --version)  installed in my laptop. I am attaching the CMake error log for your reference. Can you please solve the issue for me as it is taking really long time for a easy implementation. 

Due to this error the azure_iot.bin file is not getting generated. 

[CMakeError.log](https://github.com/azure-rtos/getting-started/files/5197157/CMakeError.log)
[CMakeOutput.log](https://github.com/azure-rtos/getting-started/files/5197159/CMakeOutput.log)

ryanwinter, 2020-09-09:
Both of those files look correct. You can ignore the CMakeError.log as this is due to the compiler check failing as its an embedded target and this is ignored by CMake.

So what files do you see in the build/app directory?

What is the console output of rebuild.bat? You should see something like:
```
-- Build files have been written to: D:/projects/iot/getting-started/STMicroelectronics/STM32L4_L4+/build
[805/810] Linking C executable app\stm32l475_azure_iot.elf
Memory region         Used Size  Region Size  %age Used
             RAM:       95304 B        96 KB     96.95%
            RAM2:          0 GB        32 KB      0.00%
           FLASH:      232200 B         1 MB     22.14%
[806/810] Linking C executable app\stm32l4S5_azure_iot.elf
Memory region         Used Size  Region Size  %age Used
             RAM:       93768 B       640 KB     14.31%
           FLASH:      232248 B         2 MB     11.07%
[807/810] cmd.exe /C "cd /D D:\projects\iot\getting-started\STMicroelectr...\9 2020-q2-update\bin\arm-none-eabi-size.exe" -B stm32l475_azure_iot.elf"
   text    data     bss     dec     hex filename
 230280    1912   93408  325600   4f7e0 stm32l475_azure_iot.elf
[808/810] cmd.exe /C "cd /D D:\projects\iot\getting-started\STMicroelectr...\9 2020-q2-update\bin\arm-none-eabi-size.exe" -B stm32l4S5_azure_iot.elf"
   text    data     bss     dec     hex filename
 230336    1912   91872  324120   4f218 stm32l4S5_azure_iot.elf
[810/810] cmd.exe /C "cd /D D:\projects\iot\getting-started\STMicroelectr...bi-objcopy.exe" -Obinary stm32l4S5_azure_iot.elf stm32l4S5_azure_iot.bin"
```

surajsridhar, 2020-09-09:
Hello, 

I am unable to see my console output as it immediately shuts down without me getting a chance to read through the output. This was not happening yesterday. I downloaded the project agian on my C Drive and this issue is happening, 

If my CMake error log file is correct, where do i find the output bin file. Beacause in the path mentioned -> build/app/output.bin , there are no files there

ryanwinter, 2020-09-09:
You can run the rebuild.bat from a command window and this will allow you to see the final output and if there are any errors.

It will look like:
[image]

surajsridhar, 2020-09-09:
[image]

Getting this error... I assume it is a CMake error... How do I fix this... My current Cmake verison is 3.18.2

ryanwinter, 2020-09-09:
When you [clone the repository](https://github.com/azure-rtos/getting-started/tree/master/STMicroelectronics/STM32L4_L4%2B#clone-the-repo-for-the-tutorial), make sure you specific the --recursive flag. Your clone is not complete.

surajsridhar, 2020-09-09:
I am not able to clone the repo... Hence I downloaded it. Is the only option to clone it?

ryanwinter, 2020-09-09:
Yes you have to clone the repository to get all the dependencies. GitHub doesn't currently provide a download option which pull in all the dependencies. 

Why are you unable to clone?

surajsridhar, 2020-09-09:
I am not sure... I get an output saying 

[image]

ryanwinter, 2020-09-09:
Unfortunately this problem goes beyond the scope of this project as your computer appears to have DNS resolution issues, or you have a broken install of Git.

My suggestion would be to make sure you install the latest version of Git as outlined [here](https://github.com/azure-rtos/getting-started/tree/master/STMicroelectronics/STM32L4_L4%2B#prerequisites).

If that doesn't work, then you would need to investigate why DNS is failing to resolve. Possible firewall or anti-virus issues.

surajsridhar, 2020-09-09:
[image]

Is this an issue from your end?

ryanwinter, 2020-09-09:
I'm unsure what that screenshot is from, but github is [currently not experiencing an outage](https://www.githubstatus.com/).

surajsridhar, 2020-09-09:
Hello, 

I fixed the git issue now. I could successfully do a recusive clone of the repo. When I am running the rebuild.bat file I am getting the following error

[image]

Should I run the batch file as an admin?

ryanwinter, 2020-09-09:
What was the DNS issue about?

This means that gcc was unable to write to your file system which may indicate you have some file system corruption. If you rerun the rebuild.bat, do you get the same issue? It doesn't need to be run as admin.

surajsridhar, 2020-09-10:
**_The port 8081 is in use. To get around with the issue, configure a custom port by setting the system environment variable 'AZURE_IOT_EXPLORER_PORT' to an available port number. **
I a getting this error command when I try to view my devices on the Azure IoT explorer. Can you let me know which port to use

ryanwinter, 2020-09-10:
Use whatever port you want that you aren't already using.

closed 2020-09-10 by ryanwinter