eclipse-threadx/getting-started #141: Callbacks not working on STM32L4 Disco

moizhraj, 2020-09-17:
I am using the unmodified getting started repo. I have an STM32L4 discovery kit. The application works fine for sending the telemetry but fails for callbacks (C2D, desired prop, twin prop) except invoking methods.

ryanwinter, 2020-09-17:
Ho @moizhraj,

The current GSG model doesn't contain and C2D commands.

For the desired property, are you using IoT Explorer to trigger an update to the telemetryInterval?

[image]

moizhraj, 2020-09-17:
@ryanwinter I am doing it all through Azure IOT Hub VS code extension, but tried with explorer as well.

[image]

ryanwinter, 2020-09-17:
Ah I see, unfortunately the Hub VSCode extension doesn't support the new Plug and Play refresh so this is not the recommended tool at this time for communicating with the device.

For the Plug and Play refresh, its still in preview so is only available in certain Hub regions. centralus is the recommended location.

moizhraj, 2020-09-17:
ok, so i tried with IoT Explorer for C2D messages but it doesn't show up on the device.

ryanwinter, 2020-09-17:
The C2D message handling is very rudimentary at this time.

I have pushed an update that will now publish the reception of the C2D message, along with the properties + message body if present. This will now be printed to the console. If you pull the latest git you should now see this printout.

ryanwinter, 2020-09-17:
You will see something like:
`Received C2D message, properties='', message='hello'`

moizhraj, 2020-09-18:
@ryanwinter I just took a fresh copy and tried to build and it fails. Can you confirm? 

[image]

moizhraj, 2020-09-18:
Thanks @ryanwinter this works. I also see that there aren't any methods for sending string values, so have added below to `azure_iot_mqtt.c`

```c
UINT azure_iot_mqtt_publish_string_telemetry(AZURE_IOT_MQTT* azure_iot_mqtt, CHAR* label, CHAR* value)
{
    CHAR mqtt_publish_topic[100];

    printf("Sending telemetry with string value\r\n");

    snprintf(mqtt_publish_topic,
        sizeof(mqtt_publish_topic),
        PUBLISH_TELEMETRY_TOPIC,
        azure_iot_mqtt->nxd_mqtt_client.nxd_mqtt_client_id);

    return mqtt_publish(azure_iot_mqtt, mqtt_publish_topic, value);
}
```

is it already available?

ryanwinter, 2020-09-21:
Yes, the guides provide a sample implementation that isn't complete. You have the right idea in that other types can be implemented by modifying the existing functions.

ryanwinter, 2020-09-21:
I'm going to close this issue as it seems you are unblocked at this time. Feel free to reopen this issue, or open a new issues if needed.

closed 2020-09-21 by ryanwinter