eclipse-threadx/getting-started #163: SNTP timeout

NOCTLJRNE, 2020-11-19:
Hi, I was following this tutorial & I was stuck at "Confirm device connection details" step 7, my console showed issues with SNTP server timeout, restarting client, & it stuck in a loop as below:
[image]
I have no idea what could be the reason, thanks for any help in advance.

ryanwinter, 2020-11-19:
Hi @NOCTLJRNE,

What device are you using? I'll try and repo first to make sure this isn't a recent regression.

NOCTLJRNE, 2020-11-19:
@ryanwinter hi, I'm using the MXCHIP IoT DevKit AZ3166

ryanwinter, 2020-11-20:
There have been changes to the SNTP code recently. Is this a new Git clone? Can you try running the following:
```
git pull origin master
git submodule update --recursive
```

And then rebuild and try again?

If isn't the issue then its possible that your network is somehow interfering with the SNTP packets. I might be possible to look at wireshark to see what the traffic is doing.

NOCTLJRNE, 2020-11-20:
thank you, I'll try it out

NOCTLJRNE, 2020-11-20:
Thank you for the hint, I connected the board to my mobile data network & the problem with SNTP server being timeout is solved, may be there is something wrong with my home Wifi network as you suggest that I haven't figured out yet.

closed 2020-11-20 by NOCTLJRNE

ryanwinter, 2020-11-20:
Glad you got it working. If you find anything out on the SNTP failures for the other network, let me know. We occasionally hit edge cases and are trying to improve the sample to work across as many configurations as possible.