eclipse-threadx/getting-started #170: Sending telemetry to IoT Hub fails after SNTP time update

Maarten-Work, 2020-12-22:
Using both MXChip and STM32 devices, I noticed the following behavior (see the attached file for logging information):

**MXChip**
  - After an hour there is a SNTP time update
  - Shortly after, the device disconnects from IoT Hub
  - This is independent from interval times.

**STM32**
- Interval time > 240 seconds
  - After sending a few telemetry messages successfully, the device gets disconnected from the IoT Hub with **0x0002000c** code. 
- Interval time < 240 seconds
  - After an hour there is a SNTP time update
  - Shortly after, the device cannot send telemetry messages anymore, but it stays connected to the IoT Hub

The behavior above was repeatable, at least for 3 different runs with different interval times.

I wonder if the difference in behavior on STM32
[TelemetryLog.txt](https://github.com/azure-rtos/getting-started/files/5731495/TelemetryLog.txt)
 with different interval times is caused by keep alive processing as [documented here](https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-mqtt-support#default-keep-alive-timeout).

I cloned the **getting_started** repo master branch on Dec 10.

ryanwinter, 2021-01-04:
I see three problems here:
1. Setting the interval >240 seconds results in a disconnect. The SDK should be sending a keep alive, so will need to investigate why this is not occurring, or not stopping a disconnect.
2. The SNTP sync results in a disconnect. I'm unsure how the time is used after the initial negotiation, but there may be an issue with the clock drifting, and then the correction is causing the disconnect.
3. There is not reconnection logic in place for when a disconnect occurs. This would be an additional feature.

Maarten-Work, 2021-01-05:
On **2** it does make sense that there would be some drifting, which would mean to somehow reset the time similar to the initial negotiation if that even makes sense.
On **3**, I think it would be great to have some reconnection logic sample code available, since this doesn't seem trivial and everybody using the SDK needs it. Could reconnection logic be integrated in the SDK? I realize that there would be a size penalty, but without reconnection logic, devices would need to restart periodically which seems more like a hack.

ryanwinter, 2021-01-05:
Looking at the MXCHIP, I'm seeing a 2-3 minute drift each hour which is ridiculous. Either the clock is not setup correctly, or their is something wrong with threadx tick counter. I suspect clock is not setup as 95MHz as the RTOS expects.

With #3, the SDK has a callback which tells you when a disconnect occurs, however its up to the application to negotiate reconnection. I'll take a look more to see the best way to do this.

Maarten-Work, 2021-01-12:
Unless I am looking at the wrong location, it seems that the callback which tells when a disconnect occurs is not exposed to the application, unlike the device twin and method callbacks. For now I simply tried restarting the system when I get telemetry errors, but I am not successful with that either. When I try to reset the Azure thread I do get an error. Here is the logic I tried to use:

```
void tx_application_define(void* first_unused_memory)
{
    systick_interval_set(TX_TIMER_TICKS_PER_SECOND);

    // Create Azure thread
    UINT status = tx_thread_create(&azure_thread,
        "Azure Thread",
        azure_thread_entry,
        0,
        azure_thread_stack,
        AZURE_THREAD_STACK_SIZE,
        AZURE_THREAD_PRIORITY,
        AZURE_THREAD_PRIORITY,
        TX_NO_TIME_SLICE,
        TX_AUTO_START);

    if (status != TX_SUCCESS)
    {
        printf("Azure IoT application failed, please restart\r\n");
    } else
    {
        tx_thread_entry_exit_notify(&azure_thread, my_entry_exit_notify);
    }
}
```
I terminate the Azure Thread when getting telemetry errors, and the tx_thread_entry_exit_notify CB is called. However, I do get an error when calling tx_thread_reset in the following code:

```
void my_entry_exit_notify(TX_THREAD *thread_ptr, UINT condition)
{
    if (condition == TX_THREAD_EXIT)
    {
        UINT status = tx_thread_reset(&azure_thread);
        printf("tx_thread_reset: (0x%04x)\r\n", status);
    }
}
```
The return value of `tx_thread_reset` equals `TX_NOT_DONE`. I guess this might have to do because I defined the callback in initialization code, but this is kind of a show stopper now.

ryanwinter, 2021-01-12:
I fixed the MXCHIP drift, the external clock was incorrect.

ryanwinter, 2021-01-12:
I think this form of reset would be unsuccessful, because I think all the other threads are still running and would fail when the main restarted thread tried to respawn them. This method would likely need a cleaner shutdown so it can be restarted.

Let me take a look at this now, I should be able to trigger a middleware connect on the disconnect message.

ryanwinter, 2021-01-12:
I have implemented some basic retry logic in this PR, can you try it out and let me know how it goes?
https://github.com/azure-rtos/getting-started/pull/177

Maarten-Work, 2021-01-13:
I tested for a longer time with MXChip/AZ3166 and it works good. I still see a disconnect every hour, but got properly reconnected to IoT Hub. I also see that the drift is gone, but see only 1 SNTP update, which used to be happening every hour as well. Further info an be found in the log file (telemetry interval at 60 seconds), see [AZ3166Log.txt](https://github.com/azure-rtos/getting-started/files/5808736/AZ3166Log.txt).

I thought this would work for STM32L4S5 as well, since the changes were in common code. However, I see a large drift after an hour and after a disconnect no reconnection to the IoT Hub, see 
[STM32Log.txt](https://github.com/azure-rtos/getting-started/files/5808763/STM32Log.txt).

I assured that I am on the same branch: 

[image]

Maarten-Work, 2021-01-13:
Just for completeness, I just tested on AZ3166 with DPS as well, and that is working properly too as you can see here: 
[AZ3166LogWithDPS.txt](https://github.com/azure-rtos/getting-started/files/5810130/AZ3166LogWithDPS.txt)

ryanwinter, 2021-01-13:
The large drift was caused because I was printing an unsigned number, and the drift was -1, I fixed this in the branch.

At this time, I believe the only outstanding issues are with the STML4 boards. These boards are running a forked version of the netxduo stack, and there are a number of deficiencies in the fork we are seeing:
1. Keep alive is not working, meaning that if no telemetry is being sent out, the device is disconnected after a few minutes.
2. The disconnect callback from the middleware (which is using the mqtt disconnect callback) doesn't work.

I will investigate a little more, but I think this will need to be surfaced back to the Azure RTOS team to fix.

Maarten-Work, 2021-05-10:
I guess we can close this issue. Running fine on MXChip, and NXP HW.

closed 2021-05-10 by Maarten-Work

ryanwinter, 2021-05-11:
Thanks Martin. There is still an outstanding issue with the STM in that netxduo doesnt provide a callback when the disconnect occurs. I'll provide this feedback to the Azure RTOS team.