eclipse-threadx/getting-started #171: Didn't start shell prompt

matsujirushi, 2021-01-02:
This repository has the devcontainer.
Open it in VSCode Remote-Container and click 'New Terminal', I cannot see the shell prompt.
[image]

ryanwinter, 2021-01-04:
Hi @matsujirushi,

I have noticed that starting a new terminal in a devcontainer can take a while. I noticed in your screenshot that you have 4 bash terminals open, have they all failed to generate a terminal?

matsujirushi, 2021-01-04:
Hi @ryanwinter ,

Thanks for question.
I waited and was prompted. about 45sec.

Each time I press ENTER, it will take some time.
It looks like oh-my-bash seems to be affected.

When I remove "export OSH=..." in ~/.bashrc, it is fine.

ryanwinter, 2021-01-08:
Hi @matsujirushi,

I'm not sure what the shell is so slow, whether this is a VSCode container issue, or something related to the getting started container itself. I'll investigate further and let you know.

TiejunMS, 2021-02-05:
One possible is that the repo is cloned in Windows file system but the container is Linux file system. Some of the shell will perform a `git status` and it takes a while. Here is the link of best practice to use docker in Windows: https://docs.docker.com/docker-for-windows/wsl/#best-practices

ryanwinter, 2021-02-05:
Thanks @TiejunMS,

Yes when I clone the repository into the within the Linux file system, then the performance issue goes away.

To do this, I chose the "Remote-Containers: Clone Repository in Container Volume..." option.

I will close this for now as this is a limitation of the WSL integration with Docker.

closed 2021-02-05 by ryanwinter