eclipse-threadx/getting-started #183: Write protected

rploeg, 2021-01-30:
Hello,

I have tried the getting started, but when I try to copy the bin file to the ST RTOS Device I received an error that the disk is protected. I am using Windows 10 (did also run as admin) The file explorer says that the disk is write protected, how can I disable that? I have tried with diskpart to do that, but that didn't the job.

rploeg, 2021-01-30:
Found the error: I had old WiFi firmware on the device (was brand new..)

closed 2021-01-30 by rploeg

ryanwinter, 2021-01-30:
Hi @rploeg, did the guide and protect help you identify the outdated firmware? We have tried to highlight this but it can be very easy to miss with all the steps, would appreciate any feedback.

rploeg, 2021-01-31:
yes, i really missed it in the manual (my fault). But when booting the device I could see in Termite the message that my WiFi firmware was out to date, so I thought let's start with that. Directly updated 5 devices :) 

@ryanwinter  maybe also something that you could help me with; I am looking for some example to use the GPIO (to enable a motor) on my ST device. Do you have some examples? In the end I want to control my motor via Azure IoT Central.

oh-domino, 2021-04-02:
@ryanwinter @rploeg How do you update the firmware on the device? There aren't any instruction for how to install the firmware in the tutorial. I've downloaded it and have the .bin and .bat files on my windows machine, but what is the best way to get them on the STM Discovery kit devices?

oh-domino, 2021-04-02:
@ryanwinter I got this working, but it took too much digging around google to figure out how to make this work. 
First, I needed to follow the readme.txt file that is in the Inventek ISM 43362 Wi-Fi module firmware update zip folder. My recommendation is that this would be better placed on the troubleshoot page of the getting started guide. 
Next, I downloaded the ST-Link Utility and the STMFlashLoader tools without any problems. 
Then the hard work really started. The ISM43362_M3G_L44_SPI_C3.5.2.5.STM SPI Firmware file does not download from the link provided. The www.inventeksys.com website has a number of issues and is frustrating to use. Makes you question their module and code. 
Fortunately, I found the binary file on the ST Community website here - https://community.st.com/s/question/0D53W000005qaNR/correct-wifi-firmware-for-bl475eiot01a-discovery-to-connect-to-aws . Once I had that, I was able to follow the rest of the readme.txt from the downloaded .zip file and update the firmware for the Wifi module.
Why doesn't the binary file come in the downloaded .zip file or at least provide it on a website that works?

ryanwinter, 2021-04-05:
Thanks @oh-domino,

I'll pass that feedback onto STMicroelectronics and I would highly recommend you provide this feedback as well (I think the forum is the only option). The Inventek website appears to be working now, but I also have had many issues with it in the past.