eclipse-threadx/getting-started #188: Controlling other devices

rploeg, 2021-02-03:
Hello,

I have the sample up and running on my MXchip with Azure RTOS, but I am looking for a good example how I can control for example a motor what I did with the MXchip for demo's. Normally I could enable pins on this way when developing for the MXchip:

if (temperature >= temperatureThreshold)
  {
    digitalWrite(PB_0, HIGH);
  }

How can I achieve this with this RTOS example?

ryanwinter, 2021-02-04:
Hi @rploeg,

The RTOS samples is built on the stm32cubef4 sdk, which is available in the GSG at [this location](https://github.com/azure-rtos/getting-started/tree/master/MXChip/AZ3166/lib/stm32cubef4/Drivers).

The HAL has function for writing to GPIO pins, which is probably of interest to you here:
https://github.com/azure-rtos/getting-started/blob/master/MXChip/AZ3166/lib/stm32cubef4/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_gpio.h

ryanwinter, 2021-02-04:
You can take a look at some of the samples over in the STM github on this topic where it shows you how to initialize the GPIO, and then toggle the pins:
https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Projects/STM32F412G-Discovery/Examples/GPIO/GPIO_IOToggle/Src/main.c

closed 2021-02-04 by ryanwinter