eclipse-threadx/getting-started #256: arm-none-eabi-gdb.exe

19valentin99, 2021-06-21:
Good morning; 
I was following the video: "Getting Started with IoT devices and Azure" (https://www.youtube.com/watch?v=MFGqlNeTWHk&list=LL&index=142&t=522s) but when I am trying to use the debugger to run the applicaiton I get the next error:
[image]

(OpenOCD GDB executable "arm-none-eabi-gdb.exe" was not found. Please configure "cortex-debug.armToolchainPath" or "cortex-debug.gdbPath" correctly)

I would like to ask if by any chance I missed something or if there are any advices you can give to me to solve this problem.
Thank you very much.

19valentin99, 2021-06-21:
I got it working by following the tutorial from github (copy-paste the binary file) but I would like to ask if there would be a sollution for this problem.
Thank you

ryanwinter, 2021-06-21:
Hi @19valentin99,

It looks like the error is because arm-non-eabi-gcc.exe is not in your system path.

Your options would be:
1. Add GDB to the path, there is a checkbox on the last page of the installer where you can do this automatically.
2. Set the GDB path in the cortex-debug settings as specified in the error.

#1 is usually the easiest option, unless you have multiple arm-none-eabi-gcc version installed.

Ryan

closed 2021-06-22 by 19valentin99