eclipse-threadx/getting-started #277: Sending telemetry to IoT Hub fails after SNTP time update issue is back

rschuerer, 2021-07-23:
Hello

The previously closed issue #170 is back.

The problem was fixed with Azure IoT middleware reconnect logic #177 
But recently this reconnect logic was removed #265 

Now the previous problem is back.

I couldn't find the reason why this fix was removed.

Regards,
Reinhard

ryanwinter, 2021-08-05:
The logic was removed because it was incomplete and was contributing to connection problems so was not a good example of how to implement reconnection logic.

If there is demand to see improved reconnection logic implemented in a sample, then we can add this to our backlog for a future feature implementation.

acomas1, 2021-08-08:
> The logic was removed because it was incomplete and was contributing to connection problems so was not a good example of how to implement reconnection logic.
>
> If there is demand to see improved [...]

Could we at least insert code to Reboot the board after a delay, when this happens?

CastleArg, 2021-09-08:
It would be great to have examples that would work for longer than 1hr!

ryanwinter, 2021-09-08:
Hi @CastleArg + @acomas1,

Is there a specific problem you are looking to solve, or are you just interested in general guidance on reconnection logic? We are investigating adding a reconnection workflow, however this does add complexity to the sample and we are worried that this may detract from the main purpose which is to demonstrate the use of the use of the Azure IoT SDK's.

Your feedback is helpful for us to decide on features for current and future samples :)

Thanks
Ryan

acomas1, 2021-09-08:
I was planning to build a Sailboat telemetry application based on the MXChip, but if the chip can't maintain a connection to Azure, I'll look for another solution.  But any guidance on reconnecting logic would be an ok work-around.

Andrew

ryanwinter, 2021-09-09:
Hi @acomas1, @CastleArg 

I have added the original reconnection logic back in, until we have time to develop a more comprehensive sample.

https://github.com/azure-rtos/getting-started/pull/292

closed 2021-09-23 by ryanwinter