eclipse-threadx/getting-started #280: Screen.draw() is offset by one

nmattia, 2021-07-24:
_Not sure if this the correct repo. The faulty code is bundled in `stm32f4/2.0.0/system/libdevkit-sdk-core-lib.a` but I can't find the source._

I noticed that when using `draw()`, the bitmap is not printed correctly. There is always a one pixel offset on the left. If the bitmap is _not_ as wide as the display, then it shows a small gap. If the bitmap is supposed to fit the display, then it will wrap around and show the _last_ column at x = 0.

Here's a picture of the issue. The horizontal line is made of underscores, showing that the problem does _not_ occur when using `print()`. However one line below, the vertical line is one pixel to the right, whereas it should be all the way on the left.

[image]

``` cpp
void setup() {
  Screen.init();
}

void loop() {
  // Print some underscores for reference
  Screen.print(1, "________________");

  // Draw a single vertical line at x = 0 below the underscores
  unsigned char bmp[1] = { 255 };
  Screen.draw(0, 4, 1, 4 + 1, bmp);
  
  // Loop forever
  while (true) {
    delay(1000);
  }
}
```

ryanwinter, 2021-08-05:
Hi @nmattia,

This repository is for connecting devices to Azure IoT using Azure RTOS. The MXCHIP uses an STMF4 and also has a small screen onboard but I'm unsure if this is the same library as this isn't part of the mbed-os.

nmattia, 2021-08-06:
Hi @ryanwinter , thanks for the reply. Any idea where I should report this issue then?

ryanwinter, 2021-08-13:
Sorry I'm not sure. The more info above mentions mbed, so perhaps the mbed forums could help.

closed 2021-08-13 by ryanwinter