eclipse-threadx/getting-started #295: [mimxrt1050] openocd config file issue

mehrdadh, 2021-09-22:
I tried to use this file 
https://github.com/azure-rtos/getting-started/blob/cd17938a9ed7325d6ac3c6b194ee6411226d0275/NXP/MIMXRT1050-EVKB/tools/mimxrt1050-evkb.cfg

But I get this error:
```
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: Debug adapter does not support any transports? Check config file order.
Error: unable to select a session transport. Can't continue.
shutdown command invoked
```
I used openocd 0.11.0.

Any idea what's the problem?

ryanwinter, 2021-09-23:
To me it's indicating that it cant find the NXP 1052 device.

Do you see a RT1050-EVK file share and more importantly, do you have a "Universal Serial Bus devices | WebUSB: CMSIS-DAP" device showing up in Device Manager?

mehrdadh, 2021-09-23:
@ryanwinter thanks for getting back to me. yeah I do see that. Here is the output of the lsusb -v:
```
Bus 002 Device 027: ID 0d28:0204 NXP LPC1768
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0d28 NXP
  idProduct          0x0204 LPC1768
  bcdDevice           10.00
  iManufacturer           1 ARM
  iProduct                2 DAPLink CMSIS-DAP
  iSerial                 3 0227000047784e45001e9004d7450038ddb1000097969900
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          130
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              7 USB_MSC
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              4 mbed Serial Port
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x03
          call management
          use DataInterface
        bDataInterface          2
      CDC ACM:
        bmCapabilities       0x06
          sends break
          line coding and serial state
      CDC Union:
        bMasterInterface        1
        bSlaveInterface         2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              32
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 
      iInterface              5 mbed Serial Port
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              6 CMSIS-DAP
```

Can you clarify which openocd version are you using? I have looked at the openocd github and I don't see any [target](https://github.com/openocd-org/openocd/) with name `imxrt1050`. So my guess is that you have a openocd branch that supports this board?

ryanwinter, 2021-09-23:
I'm running a version of openocd from over a year ago on Windows, its listed as 0.10.0. I just tested this out and it appears to not be working for me either, I'll have to investigate what is going on.

Before you go down this route though, what project are you trying to use? Even if you get the openocd interface working, its only capable of programming the internal ROM which is only 96KB so unless your application is really small, I don't think its going to fit.

mehrdadh, 2021-09-23:
@ryanwinter thanks for checking. I'm working with MIMXRT1050-EVKB board. I actually don't necessarily need to use openocd, I'm mostly looking for a way to flash my ELF file to hyperflash without drag and drop to mounted drive. One way is to mount the drive and copy to it, but is there a better way to do that? using JLink or something like that.