eclipse-threadx/getting-started #307: Connection lost after 1h

cburdea, 2021-11-16:
I used the given documentation to connect my DevKit to IoT Central. 
https://docs.microsoft.com/en-us/azure/iot-develop/quickstart-devkit-mxchip-az3166
Following the tutorial, my connection gets lost after one hour, I assume it is due to the SAS token renewal? 
Any quick fix to this?

I have appended the Termite log:
[image]

ryanwinter, 2021-11-16:
Hi @cburdea,

Yes you are correct, the disconnect is happening b4ecause of SAS token expiration. There is some preliminary reconnect logic in the sample, however it is pretty poor. 

There is a new reconnect workflow in development, can you give that a try? Hopefully this will be pushed up in a few weeks once its gone through further validation.

https://github.com/azure-rtos/getting-started/pull/302

cburdea, 2021-11-16:
Works like a charm @ryanwinter! Thanks

closed 2021-11-16 by cburdea