eclipse-threadx/getting-started #308: Add support for STM32H735G-DK

escherstair, 2021-11-18:
Can you add the getting started procedure for [STM32H735G-DK](https://www.st.com/en/evaluation-tools/stm32h735g-dk.html)?

Thanks

ryanwinter, 2021-11-19:
hi @escherstair,

We are not looking at expand our cmake guides to include additional components at this time.

However you could take a look at the STM32F746G-DISCO samples `[here](https://github.com/azure-rtos/samples)`, it is also a cortex chip and contains the same ethernet controller it could be a good starting point.

There are also [ready to run examples](https://www.st.com/content/st_com/en/products/embedded-software/mcu-mpu-embedded-software/stm32-embedded-software/stm32cube-expansion-packages/x-cube-azrtos-h7.html) available from ST for Azure RTOS that another place to look.

Ryan

escherstair, 2021-11-22:
Hi @ryanwinter 
this repo is really cool because it hosts cmak examples that can be used out-of-the-box with VSCode.
All the other links you provided are related to some specific IDE (IAR, STMCube, ...).
Do you know any tool to convert these kind of project files to cmake?

ryanwinter, 2021-11-25:
Hi,

The Cmake is created by hand and as far as I'm aware, there isnt any good tool to do this automatically from an IDE. Even if there was, Im guessing the cmake would end up unreadable. We spent some time trying to create a cmake structure that used modern cmake and was reasonably easy to understand.

To create a new one for the board above, you would probably need to the following steps by modifying the stm32l4 cmake project:
1. Use to [stm32cubef4](https://github.com/STMicroelectronics/STM32CubeF4) to replace the stm32cubel4 in `/lib/stm32cubel4`
2. Find the required startup and linker files and replace these in the `app/startup/gnu` directory.
3. Edit the tx_initialize_low_level.S in the `app/startup/gnu` directory to contain the correct system clock (L51)
4. Replace the wifi driver with the ethernet driver in `/lib/netx_driver`
5. Configure netxduo in the nx_user.h in the `app/lib/netxduo` directory. This may just be removing the `NX_ENABLE_TCPIP_OFFLOAD` and `NX_ENABLE_INTERFACE_CAPABILITY` as these are specific to the Wifi driver on the L4
6. Update the board_ini.c in `app` as well as the files in `app/stm32cubel4' which contain the board initialization code as well as interrupt redirection code.

I know that might seem like alot, and it is. Unfortunately there are many little pieces we that need to be lined up to bring together MCU along with threadx/netxduo/azure iot/etc, but I hope this helps as a starting point.

closed 2021-12-02 by ryanwinter