eclipse-threadx/getting-started #338: no tools for macos

chrisvugrinec, 2022-06-09:
Is it possible to have the instructions and tooling available for macos. Currently there are only scripts for getting the tools for windows and linux

ryanwinter, 2022-06-09:
Hi @chrisvugrinec,

Just to clarify, are you referring to the guides [here](https://docs.microsoft.com/en-us/azure/iot-develop/quickstart-devkit-mxchip-az3166)?

We haven't had a big signal for MacOS users, and in reality there are very few actual windows dependencies, really just Termite for the serial terminal, I'm not sure what the best tool would be.

Is there anywhere specific you are stuck? I'm just wondering if you are looking for help to install the required tools, building the application, flashing the device or running the terminal (or all of the above!)

ryanwinter, 2022-06-28:
Closing for now. If you want to reengage on this topic, please reopen this issue or open a new one.

campbellcole, 2022-11-22:
I am using macOS for this project and I was able to adapt everything to homebrew commands without a problem, but I can't figure out how to successfully build the project. The build script works on macOS but it fails because "warnings being treated as compiler errors". I tried modifying the build cmake file and removing the `-Werror -Wall` flags but that doesn't seem to change anything. The console output is absolutely massive and I don't think it's very helpful, but I can try to get it on pastebin if you think it would help. Thanks in advance.

ryanwinter, 2022-12-02:
What version of arm-none-eabi-gcc are you using?

ryanwinter, 2022-12-02:
Definitly a pastebin or something would be useful. You can definitly turn off the -Werror flag by editing the toolchain file: https://github.com/azure-rtos/getting-started/blob/master/cmake/arm-gcc-cortex-toolchain.cmake

ryanwinter, 2022-12-12:
Hi @campbellcole, where you able to resolve this?

ryanwinter, 2023-01-18:
Just pinging on this before I close it.

I'm wondering if you are using a newer version of the arm compiler than the one we use in our samples which might explain why it didnt compile cleanly.