eclipse-threadx/getting-started #343: Different results with two CMake projects for same sources

navinreddy23, 2022-06-27:
Hi,

I am facing a weird issue. I have made two projects with same source files that use CMake for compilation.

1. First project has CMakeLists.txt in multiple subdirectories.
2. Second project has a single CMakeLists.txt

Both projects using the same tool chain and compilation flags. But the first project hard faults when trying to execute `_tx_thread_schedule();`. The second project works without any issues. 

I have followed the project structures that are similar to the examples in this repo. Which is that, I create libraries for threadX, netxDuo and STM32Hal. 

I am wondering what could have gone wrong.

I would be happy to share the sources.

Device: STM32H723
Example project: TCP Echo Client

For reference, please take a look at the image below for the debug trace. Note that the WWDG is not enabled in the project.
[image]

ryanwinter, 2022-06-27:
Hmm, its hard to see what the problem is here, the fact that it links should mean that symbol is available but if it cant resolve this at run 

Where you able to set a breakpoint in _tx_initialize_kernel_enter and try and step into _tx_thread_schedule?

If you are happy to share the source, I can take a look. CMake can be tricky especially with the some of the optimisation flags which can drop pieces of code that it doesnt think are needed.

navinreddy23, 2022-06-28:
Hi @ryanwinter,

Thank you for the quick response!

> Where you able to set a breakpoint in _tx_initialize_kernel_enter and try and step into _tx_thread_schedule?
I am not sure at the moment, but I think I had issues when I set a breakpoint in `_tx_initialize_kernel_enter`. I can confirm this tomorrow, as I do not have the device right now.

Please find the sources at this link below. I have uploaded both the projects.
[SourcesLink](https://www.dropbox.com/s/ch01wuchb0zif1v/GitHub.7z?dl=0)

ryanwinter, 2022-06-30:
Hi @navinreddy23,

I dont have an STM32H7 board available so I modified the project to be a generic Cortex M4 project and was able to execute successfully (start executing a thread within ThreadX).

Did you have any success debugging further? Also in the screenshot above, is there a line number assigned to the end of the _tx_initialize_kernel_enter call stack line?

navinreddy23, 2022-06-30:
Hi @ryanwinter,

> Did you have any success debugging further? Also in the screenshot above, is there a line number assigned to the end of the _tx_initialize_kernel_enter call stack line?'

I tried debugging with the break point set at `_tx_thread_schedule();` at line number 142. I am able to get into `tx_thread_schedule.S`. The last point of execution is at line 102 where interrupts are enabled. 
````
    /* Enable interrupts */
    CPSIE   i
````

[image]

> I dont have an STM32H7 board available so I modified the project to be a generic Cortex M4 project and was able to execute successfully (start executing a thread within ThreadX).

I tried adding another simple thread in `UINT App_ThreadX_Init(VOID *memory_ptr);` initialization function. This didn't seem to work either with the Lib type build. But works with the single CMakeLists.txt type project.

navinreddy23, 2022-06-30:
Hi Ryan,

Wow! Thanks a lot for the explanation! 

I modified the add_library() function as you suggested, and it WORKS well!

Thank you so much for your time and helping me out!

````
20:07:32  -  [INFO]    :  [Serial]  Ethernet Lib Test OBJECT...
20:07:35  -  [INFO]    :  [Serial]  
20:07:35  -  [INFO]    :  [Serial]  STM32 IpAddress: 192.168.1.180 
20:07:35  -  [INFO]    :  [Serial]  
20:07:35  -  [INFO]    :  [Serial]  UDP Thread Started
20:07:35  -  [INFO]    :  [Serial]  
20:07:35  -  [INFO]    :  [Serial]  TCP Thread Started
20:07:59  -  [INFO]    :  [Serial]  
20:07:59  -  [INFO]    :  [Serial]  -------------------------------------
20:07:59  -  [INFO]    :  [Serial]  	TCP SUCCESS : 50000 / 50000 packets sent
20:07:59  -  [INFO]    :  [Serial]  -------------------------------------
20:08:11  -  [INFO]    :  [Serial]  
20:08:11  -  [INFO]    :  [Serial]  -------------------------------------
20:08:11  -  [INFO]    :  [Serial]  	UDP SUCCESS : 50000 / 50000 packets sent
20:08:11  -  [INFO]    :  [Serial]  -------------------------------------
````

ryanwinter, 2022-06-30:
Glad this is resolved! I havent definitly run into this problem in the past and its so hard to debug!

I have this as a pending blog topic, so I think while this is fresh in my mind I'm going to write this.

closed 2022-06-30 by ryanwinter

navinreddy23, 2022-07-14:
Hi @ryanwinter,

There seems to be another issue with the Lib type build.

It works fine when there are no interrupt handlers in the `App` directory. If an interrupt handler or an ISR callback is present, then I get a **Stack Overflow** on one of the threads (only one user thread at the moment).

If there is no interrupt handler in the `App` folder, then a Stack Overflow is not triggered.

Also, like before, if I use a single CMakeLists.txt file, then I don't get a Stack Overflow either.

Please let me know if you would like to take a look at the sources. I can share them privately.

Below is how I have structured the `App` folder `CMakeLists.txt`. 

````
cmake_minimum_required(VERSION 3.21 FATAL_ERROR)

include(${CMAKE_SOURCE_DIR}/cmake/arm-gcc-cortex-m7.cmake)
# Set up the project
#project(App
#    LANGUAGES C CXX ASM
#)

set(SYS_SRC
        SYS/OS/OS.c
        SYS/OS/OS_Queue.c
        SYS/Network/Network.c
        SYS/Network/Network_DHCP.c
        SYS/CLI/CLI.c
        )

set(HAL_SRC
        HAL/HAL.c
        HAL/Serial/Serial.c
        )

set(MAIN_SOURCES
        ${CMAKE_SOURCE_DIR}/CubeSources/STM32CubeIDE/Application/User/syscalls.c
        ${CMAKE_SOURCE_DIR}/CubeSources/STM32CubeIDE/Application/User/sysmem.c
        ${CMAKE_SOURCE_DIR}/CubeSources/Src/main.c
        ${CMAKE_SOURCE_DIR}/CubeSources/Src/tx_initialize_low_level.S
        ${CMAKE_SOURCE_DIR}/CubeSources/STM32CubeIDE/Application/Startup/startup_stm32h723zgtx.s
        )

add_executable(${PROJECT_NAME}
        ${SYS_SRC}
        ${HAL_SRC}
        ${MAIN_SOURCES}
        ${LINKER_SCRIPT}
        )

set_target_linker(${PROJECT_NAME} ${CMAKE_SOURCE_DIR}/CubeSources/STM32CubeIDE/STM32H723ZGTX_FLASH.ld)

target_link_libraries(${PROJECT_NAME} PUBLIC
        threadx
        netxduo
        stm32h7_hal
        netx_driver
        )

post_build(${PROJECT_NAME})

target_include_directories(${PROJECT_NAME} PUBLIC
        SYS/Inc
        HAL/Inc
        ${CMAKE_SOURCE_DIR}/CubeSources/Inc
        )

target_compile_definitions(${PROJECT_NAME} PUBLIC
        CONFIG_PRINT_FILENAME_FN_LINE_NR=0
        )

target_link_directories(${PROJECT_NAME}
        PUBLIC
        ${CMAKE_SOURCE_DIR}/CubeSources/Src
        ${CMAKE_SOURCE_DIR}/CubeSources/STM32CubeIDE/
        ${CMAKE_SOURCE_DIR}/CubeSources/STM32CubeIDE/Application/Startup
        )
````