eclipse-threadx/getting-started #348: IOT_DPS_ID_SCOPE empty azure error at connection

sergiosolorzano, 2022-09-01:
Hi, I can connect to wifi when I flash mxchip but I get this error at connection.

ERROR: azure_iot_nx_client_dps_entry incorrect parameters as shown in Termite:

`Initializing WiFi
	MAC address: XX:XX:XX:XX:XX:XX
SUCCESS: WiFi initialized

Connecting WiFi
	Connecting to SSID 'MYSSID'
	Attempt 1...
SUCCESS: WiFi connected

Initializing DHCP
	IP address: XXX.XX.XX.X
	Mask: 255.255.255.XXX
	Gateway: XXX.XX.XX.X
SUCCESS: DHCP initialized

Initializing DNS client
	DNS address: XXX.XX.XX.X
SUCCESS: DNS client initialized

Initializing SNTP time sync
	SNTP server 0.pool.ntp.org
	SNTP server 1.pool.ntp.org
	SNTP time update: Sep 1, 2022 9:21:51.41 UTC 
SUCCESS: SNTP initialized
ERROR: azure_iot_nx_client_dps_entry incorrect parameters

Initializing DHCP
	IP address: XXX.XX.XX.X
	Mask: 255.255.255.240
	Gateway: XXX.XX.XX.X
SUCCESS: DHCP initialized

Initializing DNS client
	DNS address: XXX.XX.XX.X
SUCCESS: DNS client initialized

Initializing SNTP time sync
	SNTP server 0.pool.ntp.org
	SNTP server 1.pool.ntp.org
	SNTP time update: Sep 1, 2022 9:22:1.253 UTC 
SUCCESS: SNTP initialized

IoT connection backoff for 15 seconds
ERROR: azure_iot_nx_client_dps_entry incorrect parameters

Initializing DHCP
	IP address: XXX.XX.XX.X
	Mask: 255.255.255.240`

I think IOT_DPS_ID_SCOPE may need to be provided in azure_config.h. However this document makes no reference to IOT_DPS_ID_SCOPE and it does not show in Azure IoT Explorer.

I see in [documentation to connect with Azure IoT Central](https://docs.microsoft.com/en-us/azure/iot-develop/quickstart-devkit-mxchip-az3166) that this param would be provided, but with Azure IoT Explorer it is not the case.

[image]

versus Azure IoT Central documentation where scope would be provided:
[image]

Please help, thank you, Sergio

[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ad78bcc1-3658-8084-4ffe-33ff562e8e72
* Version Independent ID: 2a24752e-bc92-43ac-c911-0af9269be285
* Content: [Connect an MXCHIP AZ3166 to Azure IoT Hub quickstart](https://docs.microsoft.com/en-us/azure/iot-develop/quickstart-devkit-mxchip-az3166-iot-hub)
* Content Source: [articles/iot-develop/quickstart-devkit-mxchip-az3166-iot-hub.md](https://github.com/MicrosoftDocs/azure-docs/blob/main/articles/iot-develop/quickstart-devkit-mxchip-az3166-iot-hub.md)
* Service: **iot-develop**
* GitHub Login: @timlt
* Microsoft Alias: **timlt**

sergiosolorzano, 2022-09-01:
Found the culprit, in case anyone runs in this issue, I found enabling diagnostics and waiting a bit after trying to connect logged there was an authentication problem. I created the SAS key y'day and it probably expired , renewing the primary key of the iot hub - in iothub ->shared access policies->iothubowner. Updated the flashed script and connected fine.

closed 2022-09-01 by sergiosolorzano

ryanwinter, 2022-09-01:
Thanks for the detail here @sergiosolorzano. We do know that debugging connectivity is hard and time consuming and we are looking to develop a diagnostic tool in the future to help here. I have added SAS token validation to our list.