eclipse-threadx/getting-started #349: mxchip press button send signal to Azure IoT

sergiosolorzano, 2022-09-13:
Can someone share an example or guidance how I can add that when a button in the mxchip IoT device is pressed the signal is sent to Azure IoT? I'm using this [example](https://github.com/azure-rtos/getting-started/tree/master/MXChip/AZ3166/app) to send devkit data to IoT hub.

I see the function in board_init.c that lights up the led:

`weak void button_a_callback()
{

WIFI_LED_ON();
AZURE_LED_ON();
USER_LED_ON();
if (BUTTON_A_IS_PRESSED)
{
val += 32;
if (val > 2047)
val = 2047;
RGB_LED_SET_R(val);
RGB_LED_SET_G(val);
RGB_LED_SET_B(val);
}
//set_led_state(true); I'd like to add something like this`
}`

`static void set_led_state(bool level)
{
if (level)
{
    printf("\tLED is turned ON\r\n");
    HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13, GPIO_PIN_SET);
}
else
{
    printf("\tLED is turned OFF\r\n");
    HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13, GPIO_PIN_RESET);
}
}`

How can I link these two functions, meaning from button_a_callback to call nx_client.c set_led_state function?

Thanks a bunch

ryanwinter, 2022-09-15:
So properties_complete_cb is static in nx_client.c and is called when the device twin properties have completed processing so I wouldnt use this function.

What I think you want to do is implement button_a_callback in nx_client.c (the version in board_init.c is weak, so any other implementation will override it).

You could have something like:

```
void button_a_callback()
{
    GPIO_PinState gpio_state = HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13);
    azure_iot_nx_client_publish_bool_property(&azure_iot_nx_client, NULL, LED_STATE_PROPERTY, gpio_state == GPIO_PIN_SET);
}
```

I havent tested the code, but I believe you should be able to read the gpio state in the current implementation. If not, then you could just cache the led_state in a global bool and just send that state.

ryanwinter, 2022-09-21:
Closing for now, please reopen if you have more questions.

closed 2022-09-21 by ryanwinter