eclipse-threadx/getting-started #350: DNS client fails to initilaize

nscooling, 2022-09-28:
I'm following along the tutorial for the MXChip

I keep getting the following error:
```
SUCCESS: WiFi connected

Initializing DHCP
        IP address: xx.xx.xx.xx
        Mask: 255.255.255.0
        Gateway: xx.xx.xx.xx
SUCCESS: DHCP initialized

Initializing DNS client
ERROR: nx_dhcp_interface_user_option_retrieve (0x00000094)
ERROR: dns_connect
```

I have tried two different WiFi's with the same result.

I cannot find any suggestions for this issue.

Any ideas?

ryanwinter, 2022-09-29:
Hi @nscooling,

The 0x94 errors resolves to NX_DHCP_NOT_BOUND, which seems to indicate that DHCP failed, however I can see that it was working in the log you posted above. Let me investigate more on why the dhcp service is returning this error.

Ryan

nscooling, 2022-10-03:
Hi @ryanwinter 

okay that seems to be starting the DHCP. It looks like it might have been an azure setting issue, but now I'm getting as far as:
```
IoT connection backoff for 65 seconds
Error: on nx_azure_iot_hub_client_initialize (0x00020002)
```
Again, I can't find any guidance to what I'm doing incorrectly. I think some it may be to do with the different README's for getting started and they are slightly out of sync with the current Azure dashboard config. So, I'm guessing it's finger trouble my end but it's not clear to me where the issue is.

Am I correct in thinking that both
```
#define IOT_DPS_ID_SCOPE        ""
```
and 
```
#define IOT_HUB_DEVICE_ID ""
```
should not be set?
I'm working from
https://learn.microsoft.com/en-gb/azure/iot-develop/quickstart-devkit-mxchip-az3166-iot-hub

nscooling, 2022-10-03:
Hi @ryanwinter 

I'm getting further now, the DNS client is now initializing okay.

I think it may have been an azure-iot config issue.

However, now I'm stuck here! 
```
Initializing DNS client
        DNS address: 192.168.1.1
SUCCESS: DNS client initialized

Initializing SNTP time sync
        SNTP server 0.pool.ntp.org
        SNTP time update: Oct 3, 2022 16:46:36.268 UTC
SUCCESS: SNTP initialized
Error: on nx_azure_iot_hub_client_initialize (0x00020002)
```
I'm guessing it's another config issue? but not a clue what I'm doing wrong!

ryanwinter, 2022-10-03:
Was this running on the branch? I only added some debugging so this shouldnt have improved things.

The 20002 is an invalid parameter error which means you are passing a null in for one of the parameters:

```
    if ((nx_azure_iot_ptr == NX_NULL) || (hub_client_ptr == NX_NULL) || (host_name == NX_NULL) ||
        (device_id == NX_NULL) || (host_name_length == 0) || (device_id_length == 0))
    {
        LogError(LogLiteralArgs("IoTHub client initialization fail: INVALID POINTER"));
        return(NX_AZURE_IOT_INVALID_PARAMETER);
    }

```

nscooling, 2022-10-04:
Thanks for all your support. Mostly finger trouble (azure-config.h) and mis-reading the getting started steps. All up and working now.

The upside is I do feel I have a better handle on the azure IoT dashboard and setup.

Apologies for not being able to follow basic instrcutions!

closed 2022-10-04 by nscooling