eclipse-threadx/getting-started #352: ERROR: Unable to find https://aka.ms/vcpkg-ce-default at command vcpkg activate

sergiosolorzano, 2022-10-14:
At step vcpkg activate I get error

Updating registry data from https://aka.ms/vcpkg-ce-default
ERROR: Unable to find project in folder (or parent folders) for undefined

PS C:\AR-VR-Github\mxchip-install\MXChip\AZ3166> vcpkg activate
warning: vcpkg-artifacts are experimental and may change at any time.
A suitable version of node was not found (required v16.15.1). Downloading portable node v16.15.1...
Downloading node...
  https://nodejs.org/dist/v16.15.1/node-v16.15.1-win-x64.7z -> C:\Users\Sergio\.vcpkg\downloads\node-v16.15.1-win-x64.7z
Extracting node...
A suitable version of cmake was not found (required v3.22.2). Downloading portable cmake v3.22.2...
Downloading cmake...
  https://github.com/Kitware/CMake/releases/download/v3.22.2/cmake-3.22.2-windows-i386.zip -> C:\Users\Sergio\.vcpkg\downloads\cmake-3.22.2-windows-i386.zip
Extracting cmake...
Downloading vcpkg-ce bundle 2022-10-12...
npm notice
npm notice New minor version of npm available! 8.11.0 -> 8.19.2
npm notice Changelog: https://github.com/npm/cli/releases/tag/v8.19.2
npm notice Run npm install -g npm@8.19.2 to update!
npm notice
Updating registry data from https://aka.ms/vcpkg-ce-default
ERROR: Unable to find project in folder (or parent folders) for undefined

Please help

Thank you!

ryanwinter, 2022-10-14:
Related to a change in vcpkg, I have submitted an issue: https://github.com/microsoft/vcpkg/issues/27241

sergiosolorzano, 2022-10-15:
Thanks a bunch @ryanwinter  , bug seems to make the tool unusable, any solution in the interim?

ryanwinter, 2022-10-15:
A workout I found would be to use the `--z-vcpkg-manifest-dir` flag to specific the directory.

` vcpkg activate --z-vcpkg-manifest-dir=C:\projects\iot\rtos\getting-started\MXChip\AZ3166`

It appears to only work with absolute paths.

sergiosolorzano, 2022-10-16:
Worked great @ryanwinter , this tool is awesome! thanks again for your help !

closed 2022-10-16 by sergiosolorzano

ryanwinter, 2022-10-18:
Checked in on this and the new version they pushed out was successfully able to activate.