eclipse-threadx/getting-started #358: DNS error

rploeg, 2022-11-27:
Hi,

I have followed the manual for the MXchip (AZ3166), but get this error when running the app on the MXchip:

`Initializing DHCP
	IP address: 192.168.1.18
	Mask: 255.255.255.0
	Gateway: 192.168.1.1
SUCCESS: DHCP initialized

Initializing DNS client
	DNS address: 192.168.1.1
	DNS address: 8.8.8.8
	DNS address: 8.8.8.8
ERROR: nx_dns_server_add (0x000000af)
ERROR: dns_connect`

My DNS is correctly working (Same Wi-Fi as other devices)

ryanwinter, 2022-11-28:
Hi @rploeg,

Let me take a look at this, maybe this is related to the duplicate DNS entries.

ryanwinter, 2022-11-29:
Please try again, fixed with https://github.com/azure-rtos/getting-started/pull/359

closed 2022-11-29 by ryanwinter