eclipse-threadx/getting-started #363: Changes to TLS certificate for Azure IoT Hub and DPS endpoints

sergiosolorzano, 2022-12-08:
Hi, [do these coming changes](https://techcommunity.microsoft.com/t5/internet-of-things-blog/azure-iot-tls-critical-changes-are-almost-here-and-why-you/ba-p/2393169) require us to update the code in this demo project? 

Thank you ! Sergio

ryanwinter, 2022-12-12:
We have already integrated the [3 root certs](https://github.com/azure-rtos/getting-started/blob/master/shared/src/azure_iot_cert.c) and you can see the additional certs be registered [here](https://github.com/azure-rtos/getting-started/blob/d3d1c1c699d3db994f68e1bebe4d555001ade5fe/shared/src/azure_iot_nx_client.c#L156).

These cert changes were merged into the sample 13 months ago, so anyone picking up the sample since that date will have the certificates.

devx247, 2025-08-07:
@ryanwinter Are you still in the project? Azure IoT Hub ends support for TLS 1.0/1.1 by 31th of August 2025. I set TLS 1.2 on Azure Hub and the IoT device didn't connect anymore. Can I set TLS1.2 in the firmware for my MXCHIP AZ3166 device? Thanks for any help

fdesbiens, 2025-08-07:
Hi @devx247.

Microsoft is not involved in the Eclipse ThreadX project anymore. Our team does not have access to Azure IoT instances.

Given this, we will likely deprecate the Azure IoT client in NetX Duo soon unless new contributors with access to Azure IoT and an interest in maintaining it step up. 

The project is alive, and we are currently preparing a release that is expected to be available by the end of Q3. However, our small team cannot support Azure IoT integration on its own, especially if Microsoft itself will not do it.

devx247, 2025-08-12:
@ryanwinter Thanks for your reply. I filed an issue maybe someone else can help: https://github.com/eclipse-threadx/getting-started/issues/410

devx247, 2025-08-12:
@fdesbiens Thanks for your reply. Sad to read that Microsoft isn't maintaining the project anymore. I filed any issue on that: https://github.com/eclipse-threadx/getting-started/issues/410