eclipse-threadx/getting-started #382: Connection to IoT hub failed

gegendy, 2023-04-05:
Hello
I tried the project with but having the below error while trying to connected to IoT hub:
*ERROR: nx_azure_iot_hub_client_connect (0x000000a3)*
PS: I had the #define ENABLE_DPS" commented out

Environment:
* Development is on visual studio 2022 enterprise edition
* The below settings was set in the config.h file:
#define IOT_HUB_HOSTNAME  "my specific IoT hub host name"
#define IOT_HUB_DEVICE_ID "STM32L4S5-001"

// ----------------------------------------------------------------------------
// Azure IoT DPS Self-Signed X509Certificate
//    Define this to connect to DPS or Iot Hub using a X509 certificate
// ----------------------------------------------------------------------------
//#define ENABLE_X509

// ----------------------------------------------------------------------------
// Azure IoT device SAS key
//    The SAS Primary key generated by Azure IoT
// ----------------------------------------------------------------------------
#define IOT_DEVICE_SAS_KEY
* 
* 
* Serial log:
========
Starting Azure thread
Initializing WiFi
             Module: ISM43362-M3G-L44-SPI
             MAC address: xx:xx:xx:xx:xx:xx
             Firmware revision: C3.5.2.7.STM
SUCCESS: WiFi initialized

Connecting WiFi
             Connecting to SSID 'xxxx'
             Attempt 1...
SUCCESS: WiFi connected

Initializing DHCP
             IP address: xxx.xxx.xxx.xxx
             Mask: 255.255.255.0
             Gateway:  xxx.xxx.xxx.xxx
SUCCESS: DHCP initialized

 Initializing DNS client
             DNS address  xxx.xxx.xxx.xxx
             DNS address 2: 8.8.8.8
SUCCESS: DNS client initialized

Initializing SNTP time sync
             SNTP server 0.pool.ntp.org
             SNTP server 1.pool.ntp.org
             SNTP server 2.pool.ntp.org
             SNTP time update: Apr 5, 2023 15:29:55.558 UTC
SUCCESS: SNTP initialized

Initializing Azure IoT Hub client
             Hub hostname: xxxxxxxxx
             Device id: STM32L4S5-001
             Model id: dtmi:azurertos:devkit:gsgstml4s5;2
**ERROR: nx_azure_iot_hub_client_connect (0x000000a3)**

TiejunMS, 2023-04-06:
@gegendy , it looks like a DNS issue. Could you run the command from your PC in the same network and paste the result? Make sure PC and device share the same router. Replace the `<DNS server>` and `<hostname>` with values from terminal output.

For Windows, open a powershell/cmd and run:
`nslookup <hostname>.azure-devices.net <DNS server>`

For Linux, open a command line and run:
`dig <hostname>.azure-devices.net @<DNS server>`

gegendy, 2023-04-06:
Hi @TiejunMS 
Here is the response from the command (I ran it on Windows 11).
Server:  xx.xxxx.net
Address:  xxx.xxx.xxx.xxx

Non-authoritative answer:
Name:    ihsu-eastus-4.eastus.cloudapp.azure.com
Address:  20.49.110.134
Aliases:  myhostname.azure-devices.net

ryantwinter, 2023-04-06:
Just to clarify, in the azure_config.h, make sure you have the IOT_HUB_HOSTNAME as the FQDM, so it would be " myhostname.azure-devices.net"

gegendy, 2023-04-06:
@timlt and @TiejunMS 
You both are brilliants!
Oh man, yes this was the issue. I fixed it now, I can now connect to IoT hub and see telemetry coming in.
Thank you so much!

timlt, 2023-04-06:
@gegendy glad you got it work!  

Thanks @TiejunMS and @ryantwinter for the suggestions/fixes.

closed 2023-04-19 by ryanwinter