eclipse-threadx/getting-started #384: Could you please help provide me some code examples about how to collect decibel using Azure-rtos?

zh0903, 2023-04-17:
We are using AZ3166 device with this repo https://github.com/SibeeshVenu/MXChip-Azure-IoT-Arduino
It can collect decibel telementry. This repo based on the DevKit [https://github.com/microsoft/devkit-sdk/releases/.](https://github.com/microsoft/devkit-sdk/releases/)

Since it didn't include Digicert G2 certificate, when we migrated to G2 certificate, the device cannot connect to IoT Hub using the old repo.
We must using Azure-rtos to re-flash our AZ3166 device. However, the repo Azure-rtos only has basic telmentry such as temperture and humidity, cannot collect decibel telementry.

Could you please help provide me some code examples about how to collect decibel using Azure-rtos?

ryanwinter, 2023-04-19:
Hi @zh0903,

Our focus is more on how to connect the device to IoT Hub rather than interface with the I2C sensors on the board, we mostly provide this as a demonstration on how to integrate the two together. The MXCHIP sample provides access for:
* hts221 (humidity)
* lps22hb (temperature)
* lis2mdl (magnetometer)
* lsm6dsl (accelerometer, gyroscope)

For a decibel reading from the microphone, the library to access this is part of the Wiced SDK, which we dont have permission to ship within this repository. You would need to find and download the SDK and utilize the library there. From my last check, the SDK is only available by installed Wiced Studio, however I believe the latest version has dropped support for this chipset so you may need to download an older version. Unfortunately this is beyond the scope of this sample for us to provide guidance.

The alternative is to update the arduino sample to use the new Digicert certificate. Although from the looks of it, the sample you listed above it shipping a precompiled version of the C-SDK so updating the cert would require a recompilation of the latest C-SDK which includes the appropriate certificate support.

closed 2023-04-19 by ryanwinter

ryanwinter, 2023-04-25:
hi @zh0903,

There is information in this issue on how to update the existing codebase with the new certificate: https://github.com/microsoft/devkit-sdk/issues/1083

zh0903, 2023-04-26:
Hi Ryan, thanks for your prompting. 
I am discussing this with Cristian.