eclipse-threadx/getting-started #385: Migrating azure-iot-developer-kit

ladeak, 2023-04-21:
I have an MXCHIP AZ3166 which previously went through of the setup steps detailed here: https://github.com/Microsoft/azure-iot-developer-kit
which now points to this repository.

 I have been able to setup the device, and it shows 'Azure IoT' on the screen, as well as I could see telemetry data in Termite, based on the step described to connect the device [here](https://learn.microsoft.com/en-us/azure/iot-develop/quickstart-devkit-mxchip-az3166-iot-hub)

I have setup my dev environment in VS Code.

But I am missing the step of re-using / adapting my existing Arduino code? Could you share some pointers?

ryanwinter, 2023-04-25:
Hi @ladeak,

This project is based on Azure RTOS, so is not compatible with Arduino / there is no migration path.

What issues are you having with the migration?

ladeak, 2023-04-25:
Just the learning curve is high, but all good.

closed 2023-04-25 by ladeak