eclipse-threadx/getting-started #386: MXChip AZ3166 Stopped connecting

driaug, 2023-05-04:
I have an MXChip which connects to an Azure IoT Central. It has been running fine for months but this week stopped connecting with the error in the Termite screenshot. I can't seem to find any great guidelines in other issues or on the web so bringing it up here as well. 

Happy to provide more information on config or credentials if required!

[image]

ryanwinter, 2023-05-08:
Looks like the following error:
`#define NX_CRYPTO_METHOD_INITIALIZATION_FAILURE  0x2000C`

This could be related to the Hub updates to the new certificate. When did you fork the repository?

ryanwinter, 2023-05-08:
Actually no, DPS isnt going through cert migration until early 2024 so this wouldn't be the issue. 

Couple of questions:

1. Are you able to connect the device directly to Hub, instead of via DPS?
2. Are you using cert or SAS key authentication?
3. Are you using individual or group enrollment?

driaug, 2023-05-10:
I'm no IoT expert so feel free to ask more clarification if these are not the answers you expect :)

1. I have not tried this yet!
2. Using the default SAS keys generated by IoT Central
3. I've created a new device in IoT Central using the existing device template which then generated two SAS keys + device ID for auth. I've tried using the existing device we've had for months and a new one, both resulting in the error above.

Somehow leaving the device on my desk without power for a couple of days seems to have fixed it, could this potentially be a rate limiting issue somewhere?

ryanwinter, 2023-05-10:
Oh ok, so using central makes sense for the DPS parts.

I don't think it would have been not working due to rate limiting, because it was blocked on the DPS component, where I think the rate limiting should be happening on the IoT Hub connection, but it's possible. Plus I don't think there are any limits on plans with Central, its' just pay as you go now.

Well if it stops working again, then let me know and we can try to debug more.

closed 2023-05-13 by driaug