eclipse-threadx/getting-started #388: DNS client initialization fail - STM32 B-L475

kae-made, 2023-06-27:
I'm trying https://github.com/azure-rtos/getting-started/tree/master/STMicroelectronics/B-L475E-IOT01A
by Visual Studio 2022.
But DSN client initialization fail happens.
Log is following

```
Initializing WiFi
        Module: ISM43362-M3G-L44-SPI
        MAC address: C4:7F:51:05:41:5A
        Firmware revision: C3.5.2.3.BETA9
ERROR: WiFi firmware is out of date
SUCCESS: WiFi initialized

Connecting WiFi
        Connecting to SSID 'my-wifi'
        Attempt 1...
SUCCESS: WiFi connected

Initializing DHCP
        IP address: 192.168.68.70
        Mask: 255.255.252.0
        Gateway: 192.168.68.1
SUCCESS: DHCP initialized

Initializing DNS client
        DNS address 1: 255.255.255.255
        DNS address 2: 255.255.255.255
ERROR: nx_dns_server_add (0x000000af)
ERROR: dns_connect
```

Please let me know how to fix this problem.

In addition, I think I need update Wi-Fi firmware. I found related site https://github.com/ARMmbed/wifi-ism43362 but my browser notices links for firmware are unsafe. I'd like to know update is necessary or not and if I should update firmware on my board please let me know safe link.

Regards,

bo-ms, 2023-06-27:
Hi @kae-made The DNS address (255.255.255.255) is incorrect, you may to upgrade the WIFI firmware, such as: C3.5.2.5.STM. You can find the new firmware here: https://www.inventeksys.com/iwin/firmware/

kae-made, 2023-06-27:
https://www.inventeksys.com/iwin/firmware/ seems to be recognized as not safe site. My web browser refuse to download firmware bin file via link on the page.

kae-made, 2023-06-30:
Hi, I success to run sample app after updated Wi-Fi firmware(ISM43362_M3G_L44_SPI_C3.5.2.7.STM.bin).
Thank you for your support.
It's ok to close this thread.

bo-ms, 2023-06-30:
Hi @kae-made good to know new firmware works.

closed 2023-06-30 by bo-ms