eclipse-threadx/getting-started #391: LED light is always off on setLedState - possible yes/no error instead of true/false options

rlone-nutissu, 2023-08-18:
The state for the setLedState - seems to be on Yes/No? I don't know how or where this needs to get changed - but the light is forever told to be turned off.

[image]

bo-ms, 2023-08-21:
Hi @rlone-nutissu, you may set the state as true/yes to turn on the yellow User LED light on the device, or set the state as false/no to turn off the yellow User LED light on the device. Thanks. Refer to https://learn.microsoft.com/en-gb/azure/iot-develop/quickstart-devkit-mxchip-az3166-iot-hub#call-a-direct-method-on-the-device

ryanwinter, 2023-08-22:
Hi @rlone-nutissu,

There is a bug in the version 0.15.8 and later of Azure IoT explorer in that it is sending yes/no instead of true/false for the boolean value in the plug and play section. The device interprets "true" as true and everything else as false.

I will raise a bug against IoT Explorer. For a workaround I would suggest uninstalling the version you have and installed version 0.15.7.

[image]

ryanwinter, 2023-08-22:
https://github.com/Azure/azure-iot-explorer/releases/tag/v0.15.7

closed 2023-08-22 by ryanwinter

ryanwinter, 2023-08-22:
https://github.com/Azure/azure-iot-explorer/issues/637